skot/bitaxe #5: BM1387 footprint pin pitch is wrong!!

skot, 2022-06-10:
rrrrrg. the pin pitch is wrong. this isn't going to work!

skot, 2022-06-11:
Switch the BM1387 pin pitch to 0.48mm in f5b4fd9. Hopefully that does it!

robomaniac, 2022-06-12:
Have you compare your footprint with this design?

https://www.pcbway.com/project/shareproject/PCB_for_Asic_BM1387__1397_.html

skot, 2022-06-13:
Maybe my gerber measuring isn't so accurate but it looks like some pins are spaced 0.5mm and some 0.46mm?

skot, 2022-06-13:
When I measure my gerbers, the tool (Cuprum) says 0.48mm for all pins. I'm not sure what's going on with this pcbway design.

robomaniac, 2022-06-13:
I just check with my gerber viewer and you are right, I also have values ranging from 0.46 to 0.5mm.
0.48mm is right in the ~middle 😅

skot, 2022-06-13:
sidehack on the bitcointalk.org forums (the guy who makes the Newpac USB miner) says he eventually landed on 0.475mm pitch.
https://bitcointalk.org/index.php?topic=5400851.msg60335480#msg60335480

skot, 2022-06-29:
Got the V2 boards back with the 0.48mm pitch. It's working well!

closed 2022-06-29 by skot