skot/bitaxe #6: core voltage pin headers are too small

skot, 2022-06-10:
The VIN pin headers used for powering the BM1387 core are a little too small and cause a pretty real voltage drop. need to come up with something different.

maybe screw terminals?

robomaniac, 2022-06-10:
Hey
This is Jerome from Reddit. Have a look at redcube from Würth Elektronik 
https://www.digikey.ca/short/9q0wpdb3

There is other alternatives. On Würth website they give footprint for major software. Not sure about kicad. 

Or even these but the circle one
https://www.digikey.ca/short/m05zbwz8

Big pad right underneath. You need screw and nut. Maybe washer and lock nut too

skot, 2022-06-10:
@robomaniac That's an excellent idea. I'll switch the VIN pins to one of those "spade terminal" style connectors today. I've got to spin these boards anyways for the pin pitch issue #5

skot, 2022-06-11:
I switched to these slotted terminals (also called spade). I think they are pretty common on industrial equipment. They're also cheap and in-stock. 
[image]

skot, 2022-06-29:
The big spade connectors are working well in V2

closed 2022-06-29 by skot