skot/bitaxe #7: Aliexpress links are dead

BastelPichi, 2022-10-14:
Alltho added yesterday, gives a 404. Just makes me wonder...

skot, 2022-10-14:
The links work for me, but I notice the "us" in the URL. Maybe it's regional?

I have not ordered or tested any of the parts from these suppliers, but I am curious. The BM1397 prices are $15 cheaper than new ICs. Supposedly they are tested and 95% working??

BastelPichi, 2022-10-14:
Interesting...
Cant find any that cheap in germany, "me sad again :-1:"

BastelPichi, 2022-10-14:
Oh yea, found a small "Not available in your area"

closed 2022-10-14 by BastelPichi

skot, 2022-10-14:
All this stuff comes from China. Maybe there is a German version of AliExpress you can search for those vendors?

BastelPichi, 2022-10-14:
> All this stuff comes from China. Maybe there is a German version of AliExpress you can search for those vendors?

I know. Somehow there arent any cheap offers on german aliexpress... Chepest is 26€.

skot, 2022-10-14:
I'm thinking about ordering a few of these used chips to test once I know this circuit works with new BM1397s. I could order a few extra to put in the mail to you if you want..

BastelPichi, 2022-10-14:
> I'm thinking about ordering a few of these used chips to test once I know this circuit works with new BM1397s. I could order a few extra to put in the mail to you if you want..

Well at least when sending letters from germany, only papaer is allowed, otherwise expensive international package shipping. But if theres cheap shipping, why not.