skot/bitaxe #12: No protection on 5V input

skot, 2022-11-25:
the 5V input powers a number of things on the board. Should it be protected somehow?

skot, 2022-11-27:
Maybe something like a PolyZen fuse; https://www.littelfuse.com/media?resourcetype=application-notes&itemid=f3e02f71-8329-4601-bbc8-e19318fb6c9f&filename=littelfuse-polyzen-device-helps-protect-portable-electronics-from-charger-induced-failure-application-note

robomaniac, 2022-11-28:
if you don't protect Vin, why protect 5V?

5V is to get 1.8V and 0.8V and then power the fan. You could protect the fan only, a PTC (Resettable Fuses) right before J4. So if someone short the J4, the fuse will trigger.

Is that really necessary? it's rabbit hole, you can add protection TVS diode (ESD protection diode) on each input, TX, RX, etc.
What you could do is add them to schematic and let the user populated them on the PCB. Place them as close to the connector, not the chip. You want to catch the ESD way before it reach the IC. Where the static discharge will be applied, aka touching with finger the J4 connector or serial port J3 of the MAX.

skot, 2022-11-28:
this issue is in regards to the new `max_v2` branch. In this new schematic VIN is 5V and it powers everything;
- 5V is switch-mode regulated down to be the core voltage for the BM1397
- 5V is LDO regulated to 3V3 for the ESP32 and some other sensors
- 5V is regulated down to 1V8 and 0V8 like you say for the IO voltages.
- 5V powers the fan

BitMaker-hub, 2023-04-02:
Hi Skot, for me is a must to use TVS diodes and PTC fuses. 
This could help a lot. There are 3A or 4A 1206 rearmable fuses really cheap.

shufps, 2023-10-23:
[image]

On my HAT I used this. But I also used +12V as input because 12V/3A power supplies are easier to get. 

If I didn't get +5V from the Pi, I would use 12V regardless and add a 5V buck because it only has advantages.

shufps, 2023-10-23:
> this looks interesting... does D1 actually get you reverse polarity protection?
> 
> @shufps are you on the OSMU discord? it might be easier to talk there. https://discord.gg/GzAg8GaT

:wave: I contacted you on discord :blush:

skot, 2024-04-25:
Maybe the [TPS25924](https://www.ti.com/lit/ds/symlink/tps25924.pdf) eFuse