skot/bitaxe #13: NebulaMiner schematic review feedback

skot, 2022-11-26:
From NebulaMiner on bitcointalk.org on 11/25/22:

Here is my Schematic review :
- I see DS4432U+ OUT0 being used for TPS40305 control, but does the OUT1 being used somewhere ? (if not use, better not creating a signal out of the package to give ambiguity...)
- where the 5V is generated ? I see +3V3 (U11), 1V8 (U5) and 0V8 (U6) being generated from 5V, and BM1397 VDD (alias of TPS40305 VOUT) being generated from VIN_M (the monitored VIN), but nowhere the 5V beoing generated...
- on the ESP, I think you inverted TX/RX, please double check it
- can you add pin header and TP for debug purpose on the 3V3 RX/TX/RST, (additionnaly to J3), and add also I2C SDA/SCL signal on this debug pin header ?
- can you use an optional (with a serial 0R resistor) signal from a CLK_OUT of the ESP (pin 13/14/34/33/32 up to you) going to the BM1397 CLKI instead of the U1 one. I may want to try later if MCU can generate this clock by SW so we have control on the hashrate of the BM1397...
- on the BM1397 you pulled down BI (with R4) are you sure it will not being used ? on BM1385 datasheet it is noted as a Busy Input (internal schmit trugger and pull down). Maybe you can link it to a GPIO of the MCU, we will see later if needed. (and add the 3V3 signal to the debug pin header too).
- for the voltage level translators (U2/U3/U4), I used to use TSX0104 in my previous design, it is easy to use because bidirectionnal. Did you consider them ?
- for EMC2101, the ALTER#/TACH pin can be configured as input (TACH monitor to mesure Fan RPM) or output (ALERT# signal to trigger an IRQ on MCU when special temp/tach conditons are met). I see more value having an ALERT# signal from it to the MCU to trigger an interrupt for Temperature Critical exceeded (with hysteresis to deassert the signal). The FAN RPM is not very usefull for us. Also EMC2101 porpose a Look-up Table of 8 stages so we can control by hardware FAN speed accoding to BM1397 internal temperature. Maybe what you can do, is link the EMC2101 ALERT#/TACH pin to a MCU GPIO, and the FAN TACH to another MCU GPIO (be careful of voltage level here), and SW will be able to configure EMC2101 in TACH reading for some seconds, by propagating the FAN signal, then switch back to ALERT interrupt for a longer period (1h ?). So most of the time we use Interrupt monitor, and only once in a while do a fAN RPM measurement for UI needs...
- it would be usedfull to have the BM1397 VDD on a ESP ADC input to measure the BM voltage
- it would be usefull to have the TPS40305 PGOOD linked to a ESP GPIO to know if Power is Good

skot, 2022-11-26:
@skot comments;

Here is my Schematic review :
- I see DS4432U+ OUT0 being used for TPS40305 control, but does the OUT1 being used somewhere ? (if not use, better not creating a signal out of the package to give ambiguity...)

The DS4432 is a dual channel part. We only need one channel. I didn't see a cheaper/better stocked current DAC alternative. The DS4432 datasheet doesn't have much to say about unused outputs. my guess is that you leave the unused channel floating and make sure not to enable it in software. (They default to off on startup)

- where the 5V is generated ? I see +3V3 (U11), 1V8 (U5) and 0V8 (U6) being generated from 5V, and BM1397 VDD (alias of TPS40305 VOUT) being generated from VIN_M (the monitored VIN), but nowhere the 5V beoing generated...

The input to the bitaxeMax is 5V. that happens at J1 and J2. There isn't any protection happening yet. Maybe that's a good idea? Any suggestions on how you would do that? I created a github issue for this. https://github.com/skot/bitaxe/issues/12

- on the ESP, I think you inverted TX/RX, please double check it

wow, oops! nice catch.

- can you add pin header and TP for debug purpose on the 3V3 RX/TX/RST, (additionnaly to J3), and add also I2C SDA/SCL signal on this debug pin header ?

sure

- can you use an optional (with a serial 0R resistor) signal from a CLK_OUT of the ESP (pin 13/14/34/33/32 up to you) going to the BM1397 CLKI instead of the U1 one. I may want to try later if MCU can generate this clock by SW so we have control on the hashrate of the BM1397...

The BM1397 has a decent PLL hashrate clock that is configurable through a couple registers. I think this is the traditional way to change the hashrate (with the corresponding core voltage change). But I agree it would be interesting to play around with this.. I'll add it.

- on the BM1397 you pulled down BI (with R4) are you sure it will not being used ? on BM1385 datasheet it is noted as a Busy Input (internal schmit trugger and pull down). Maybe you can link it to a GPIO of the MCU, we will see later if needed. (and add the 3V3 signal to the debug pin header too).

I'm pretty sure it's not needed, at least on a single ASIC chain. But if we switch to the TSX0104 level shifter, there is a spare channel, so I can hook this up. 

- for the voltage level translators (U2/U3/U4), I used to use TSX0104 in my previous design, it is easy to use because bidirectionnal. Did you consider them ?

I initially avoided it because I remember it being kinda slow. But now that I'm looking at the datasheet I see that's for open-drain mode (like for I2C). In push-pull mode it's 24Mbps, which should be fine for us. I see it's in stock too -- I'll switch to it. thanks for the reccomendation

- for EMC2101, the ALTER#/TACH pin can be configured as input (TACH monitor to mesure Fan RPM) or output (ALERT# signal to trigger an IRQ on MCU when special temp/tach conditons are met). I see more value having an ALERT# signal from it to the MCU to trigger an interrupt for Temperature Critical exceeded (with hysteresis to deassert the signal). The FAN RPM is not very usefull for us. Also EMC2101 porpose a Look-up Table of 8 stages so we can control by hardware FAN speed accoding to BM1397 internal temperature. Maybe what you can do, is link the EMC2101 ALERT#/TACH pin to a MCU GPIO, and the FAN TACH to another MCU GPIO (be careful of voltage level here), and SW will be able to configure EMC2101 in TACH reading for some seconds, by propagating the FAN signal, then switch back to ALERT interrupt for a longer period (1h ?). So most of the time we use Interrupt monitor, and only once in a while do a fAN RPM measurement for UI needs...

I feel like knowing the fan RPM is important, at the very least for feedback -- is the fan going the speed we set? And for doing a sanity check on the fan -- is it connected and working? I see what you are saying about the ALERT function though. I'll hook it up to a MCU GPIO with a voltage divider.

- it would be usedfull to have the BM1397 VDD on a ESP ADC input to measure the BM voltage

good idea, i'll add that

- it would be usefull to have the TPS40305 PGOOD linked to a ESP GPIO to know if Power is Good

good idea, i'll add that

skot, 2022-11-26:
most of these have been addressed in 1a5a904 and 8d23cc9

rapsacw, 2022-12-07:
You are working on a similar idea as I am, looking around your project motivated me to publish my work so far, maybe you can find something useful? https://github.com/rapsacw/AsiNine-Pyramid

closed 2023-10-11 by skot