skot/bitaxe #14: No drill layer?

WillR06, 2022-12-06:
I'm trying to print of the board through jlcpcb and they're saying there is no. drill layer for the board?

skot, 2022-12-06:
It's possible I forgot to generate the drill file -- which branch are you using?

WillR06, 2022-12-06:
Master branch

WillR06, 2022-12-06:
Also would it be possible for you to add just the gerber files somewhere so it's a simple process to get the boards printed on somewhere like jlcpcb?

skot, 2022-12-06:
> Also do you have any alternatives i can use instead of the "2388-PCT330/10DM-ND" capacitors? they're 330uF 10v 20% and they're out of stock on digi key but they cost £50 just to ship 10 which is a bit much for a small amount of them?

Can you make a new issue for this? I'll try and come up with some replacement ideas

robomaniac, 2022-12-06:
> > Also would it be possible for you to add just the gerber files somewhere so it's a simple process to get the boards printed on somewhere like jlcpcb?
> 
> I've been thinking about this a bit; I think it's probably best not to distribute the manufacturing files (gerbers, drill files, centroid, etc), especially for a very early stage project like this. It's kinda like distributing the binaries on a SW project. There are so many different PCB/A vendors with different requirements, that it's probably best to generate them yourself. Luckily, it's really easy with KiCad. I'll put together a short how to and add it to this repo.

I sort of disagree with that statement. If your design use standard holes and process it should be good to be manufacturer by different fab house. If you make very small trace and holes that certain fab house can only do then yes you are right.
I would suggest you post gerbers and the place your ordered from.  I don't use kicad so I can't generate gerbers. I would rather upload gerbers and quickly test this PCB before I decide to make my own PCB with my needs.

skot, 2022-12-06:
> I sort of disagree with that statement. If your design use standard holes and process it should be good to be manufacturer by different fab house. If you make very small trace and holes that certain [...]

I'm willing to revisit this when the design settles a bit. Right now, in the early stages, you should definitely open up the project and understand what you're getting yourself into before ordering PCBs.

I highly recommend checking out KiCad. It's gotten significantly better since version 6. It's also free, installs easy, and works on Mac, Linux and PC. The OSHW world has pretty much all moved over to KiCad from Eagle.

closed 2023-01-27 by skot