skot/bitaxe #15: Need some alternatives for parts

WillR06, 2022-12-07:
In need of some alternatives for the part listed as "2388-PCT330/10DM-ND" as they're out of stock on Digi key and will take almost 2 months with £50 shipping for only 10. Is there any capacitors that I can order instead that will work without needing to adjust the pcbs?

WillR06, 2022-12-08:
would. "TAJE337M010RNJ" on arrow work as a substitute?

WillR06, 2022-12-08:
its 1ohm instead of 0.9 ohm which is the only difference

skot, 2022-12-19:
Hey, sorry for the late response. `TAJE337M010RNJ` should work fine!

10V rating is probably overkill too. I bet the cheaper, 6.3V `TAJD337K006RNJ` would also work.

robomaniac, 2022-12-20:
I suggest you read about "capacitor derating". I believe the same effect is found on tantalum capacitor. Murata does have great tool (SimSurfing) to see the effect of derating or DC bias.

in nutshell the capacity of the capacitor rapidly declines as the DC voltage rises. So if you apply 5V on the 6.3V you won't get 330uF but sometimes half that value! So going to 10V and more is good idea.

in your case this is decoupling but some case it's a filter thus have greater impact of on design.

WillR06, 2022-12-21:
I was wondering if "TAJD337M010RNJ" would also work, sure its more expensive but its pretty much the exact same part but works out cheaper than the original part