skot/bitaxe #17: JTAG connector problem for ESP32-S3-WROOM-1

barnix2001, 2023-01-08:
Hi! Bitaxe is a great project. Something is wrong about JTAG for ESP32-S3-WROOM-1.
Your JTAG on PCB is: GPIO9(17), GPIO12(20), GPIO13(21), GPIO14(22), GPIO21(23), GPIO47(24)
JTAG for ESP32-S3-WROOM-1 in doc is: IO42(TMS)(35),IO41(TDI)(34),IO40(TDO)(33),IO39(TCK)(32)

barnix2001, 2023-01-08:
I do not understand the scope of this connector. Maybe if you see the Lolin S3 PRO Devkit for ESP-S3-WROOM-1 you can place a connector for TF micro SD card or a TFT display.
[See Image](https://i.ibb.co/3SBHHNx/esp32c3-1.png)

skot, 2023-01-10:
That connector is just to break out some GPIO for experimenting. I didn't have a specific plan for it.

Breaking out JTAG is a good idea though; I'll add that on the next rev!

barnix2001, 2023-01-21:
If you need braiins open souce files (bm1387.rs, axi_bm13xx_tb.sv) you can download from this link:
[https://drive.google.com/drive/folders/1gXt9nkUYpodYeJaFKOCNf2Z-g28HXuWY?usp=sharing](https://drive.google.com/drive/folders/1gXt9nkUYpodYeJaFKOCNf2Z-g28HXuWY?usp=sharing)

skot, 2023-01-21:
Ooo, yes I would like those files. @barnix2001 the directory you posted is empty for me

barnix2001, 2023-01-21:
Send me your e-amil address from google and I will send the link from google drive.
My email is: barnix2001@gmail.com

barnix2001, 2023-01-21:
Try this link: [https://uploadnow.io/f/7LsPxM3](https://uploadnow.io/f/7LsPxM3)

skot, 2023-01-21:
got it, thanks!!

skot, 2023-01-27:
I'm going to close this issue for now. If anyone stumbles upon this issue looking for the braiins open source files, check this; https://github.com/skot/braiins

closed 2023-01-27 by skot