skot/bitaxe #20: TPS40305 feedback resistors are not right

skot, 2023-01-17:
The TPS40305 voltage regulator feedback resistors R14 and R15 are not the right values.

The good news is that this makes the TPS40305 output like 0.64V which is harmless. But you'll never be able to to adjust it up to the 1.5V it needs to be.

I'll add the correct values here soon.

skot, 2023-01-17:
For a default TPS40305 output of 1.5V the feedback resistor values should be;
R14 = 4.99kΩ
R15 = 3.32kΩ

I don't have any 4.99k resistors to try yet, so I used 4.75k for R14 and got an output of 1.452V -- not bad!

developeralgo8888, 2023-01-23:
i will have to rectify this on my prototype boards before i can continue with tests

skot, 2023-05-23:
this was fixed in 1881967

closed 2023-05-23 by skot