skot/bitaxe #24: Support for Scrypt mining chips. (BM1489)

IdotMaster1, 2023-01-28:
Will bitaxe ever support scrypt chips like the BM1489? I would really like to see a scrypt version of this!\]

skot, 2023-01-29:
I don't really know anything about the BM1489 or scrypt, so prolly not anytime soon. Unless you want to fork it and give it a shot!

closed 2023-03-03 by skot

ffrediani, 2024-05-16:
@skot what would take to understand the BM1489 ASIC and make a scrypt version of Bitaxe ? Given the ASIC looks like the BM1366 physically, do you think it would have a too much different pin configuration ?

Other than that would also require any significant changes on the ESP32 code ?

Since BM1366 is an undocumented ASIC what would take to get the necessary information from it as it was for BM1366 ?