skot/bitaxe #32: 3V3 debug header silkscreen labels are from the MCU perspective

skot, 2023-03-19:
[image]

- RX is connected to the ESP32 RX
- TX is connected to the ESP32 TX

so if you are trying to communicate with the BM1397 with a FTDI, be aware that you need to connect the FTDI TX to the pin labeled TX (and same with RX)

a-1500, 2023-03-19:
Is it possible to communicate and mine on the bm1397 with cgminer without esp-miner

a-1500, 2023-03-20:
Hello, Is it possible to communicate and mine on the bm1397 with cgminer with a PC directly and could you please tell me how to flash esp-miner on the bord

skot, 2023-03-20:
yes, it'd definitely possible. The way I do it at the moment is a bit hacky. I'm working on a better solution, which I will post here soon!

some tips if you want to start looking into it in the mean time;
- You don't need ESP-Miner on the ESP32 if you're going to use CGMiner on a PC.
- You do need a FTDI usbserial adapter that does 1.8V logic _and_ has the CBUS1 pin broken out.

a-1500, 2023-03-20:
How can I use is it with an the esp32 if it's possible

a-1500, 2023-03-20:
Hello, How can I use is it with an the esp32 if it's possible

skot, 2023-03-20:
You'll need some miner firmware for the ESP32. The [ESP-Miner project](https://github.com/skot/ESP-Miner) aims to be this, but it's not functional, yet.

a-1500, 2023-03-20:
> You'll need some miner firmware for the ESP32. The [ESP-Miner project](https://github.com/skot/ESP-Miner) aims to be this, but it's not functional, yet.

How do you do it?