skot/bitaxe #42: TXB0104 does not like the 1K RST_N pulldown R1

skot, 2023-05-11:
The [TXB0104 datasheet](https://www.ti.com/lit/ds/symlink/txb0104.pdf?HQS=dis-dk-null-digikeymode-dsf-pf-null-wwe&ts=1669420543825&ref_url=https%253A%252F%252Fwww.ti.com%252Fgeneral%252Fdocs%252Fsuppproductinfo.tsp%253FdistId%253D10%2526gotoUrl%253Dhttps%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Ftxb0104) says;
> If pullup or pulldown resistors are connected externally to the data I/Os, their values must be kept higher than 50 kΩ to ensure that they do not contend with the output drivers of the TXB0104 device.

This 1K pulldown causes all sorts of oscillations. Probably related to issue #35 
[image]

skot, 2023-05-11:
The workaround for now is just to remove R1

rapsacw, 2023-05-20:
Always place all pull-downs/ups on the input side of the TXB0104, and if possible connect OE to VccA so all input states are always replicated on the output side. Pull-ups/downs are only needed for signals that are in high impedance state (during startup), and are mostly signals from your controller that are not yet configured (and maybe 1 or two from the asic).

skot, 2023-05-23:
the TXB0104 is bi-directional, so it doesn't technically have an "input side". does that matter?

rapsacw, 2023-05-23:
Heh, I doubt you use it bi-directionally (in the true sense as in switching the directionality as needed).
Anyway, place the resistors on the side that has the origin of the signal.
A little off-topic, how are things going? My project is moving slowly because I can't make up my mind on how fast the miner should be. I'm trying to stick to the max. 10-15W power consumption but I'm tempted to also make a 1THs+ version. On the software side; my esp-miner is running stable (bm1387 & bm1397) but lacks in the UI department. I think I'm going to stick with the temporary solution I've hacked up that uses a command prompt interface with some simple commands to write/type/copy/delete config files and a few shortcuts to manipulate voltage & frequency while running, all based on the WebserialLite library. I'll update the repository when I'm finished implementing the configs for pool and wifi settings.
On the BM1397 side of things; VDDIO1.8 & VDDPLL0.8 together consume only 20mA. This means a simple resistor divider can create the 0.8V for the PLL (tested and proved working using 22 & 20 ohm resistors), and it is possible (and simplest/cheapest) to use something like the mcp16251 boost converter for all asics as it can run on input voltages of below 1V (so less than Vcore) and can output 1.8V.

closed 2023-10-11 by skot