skot/bitaxe #47: Noob Question

StarGateMiner, 2023-06-11:
I came across the project, but not fully understanding if right now its something a every day person can replicate, I was looking at the guide, can buy the stuff from Aliexpress but missing the board, are you planning to make and sell the boards. Just some background would be good, thanks..

DobQA, 2023-06-15:
@skot

ralphdesktop, 2023-06-15:
I am not Skot but I think I can answer to your question. Everyone could replicate this great project, but you have to buy equipment + to take some time to learn about soldering basics and to understand how it properly works. About the board, you have to contact a PCB manufacturer yourself. More details about that here : https://github.com/ralphdesktop/bitaxe/blob/master/building.md

Have a nice day

closed 2023-10-11 by skot