skot/bitaxe #55: ESP32-C6

Lysnyt, 2023-08-05:
Hello Skot, would be not better to use the C6 version of ESP32 instead od dual core S3.
C6 have Wifi 6 implemeted. Which is not faster but should be more reliable if there are many bitaxes connected at once.

C6:
High performance 32-bit [RISC-V](https://en.wikipedia.org/wiki/RISC-V) CPU, up to 160 MHz,[[12]](https://en.wikipedia.org/wiki/ESP32#cite_note-12) implementing RV32IMAC
    Low Power 32-bit RISC-V CPU, up to 20 MHz, implementing RV32IMAC
    512 KiB SRAM and 320 KiB ROM
    [IEEE 802.11ax](https://en.wikipedia.org/wiki/IEEE_802.11ax) (Wi-Fi 6) on 2.4 GHz, supporting 20 MHz bandwidth in 11ax mode, 20 or 40 MHz bandwidth in 11b/g/n mode
    [IEEE 802.15.4](https://en.wikipedia.org/wiki/IEEE_802.15.4) ([Thread](https://en.wikipedia.org/wiki/Thread_(network_protocol)) + [Zigbee](https://en.wikipedia.org/wiki/Zigbee))
    Bluetooth 5 (LE)
    30 (QFN40) / 22 (QFN32) programmable GPIOs

S3:
Dual-core Xtensa LX7 CPU, up to 240 MHz,[[10]](https://en.wikipedia.org/wiki/ESP32#cite_note-10) and supporting [single-precision](https://en.wikipedia.org/wiki/Single-precision) [FPU](https://en.wikipedia.org/wiki/Floating-point_unit)
        Added instructions to accelerate machine learning applications
    512 KiB SRAM, 384 KiB ROM, and 16 KiB RTC SRAM
    Capable of connecting to external [PSRAM](https://en.wikipedia.org/wiki/PSRAM) and [Flash](https://en.wikipedia.org/wiki/Flash_Memory) via [Quad SPI](https://en.wikipedia.org/wiki/Quad_SPI) or [Octal SPI](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface), and share the same 32 MiB [address space](https://en.wikipedia.org/wiki/Address_space)
    Ultra-low power [RISC-V](https://en.wikipedia.org/wiki/RISC-V) (RV32IMC) coprocessor clocked at 17.5 MHz approximately
    Ultra-low power [FSM](https://en.wikipedia.org/wiki/Finite_State_Machine) coprocessor similar to previous ESP32 and [ESP32-S2](https://en.wikipedia.org/wiki/ESP32-S2)
    WiFi 2.4 GHz (IEEE 802.11 b/g/n)[[11]](https://en.wikipedia.org/wiki/ESP32#cite_note-11)
    Bluetooth 5 (LE)
    44 programmable GPIOs
    2 × 12-bit SAR ADCs, up to 20 channels
    [USB OTG](https://en.wikipedia.org/wiki/USB_On-The-Go)

skot, 2023-08-05:
Yes, definitely! I would like to switch ASAP.

Is there a similar C6 module?

Lysnyt, 2023-08-05:
In your BOM list is ESP32-S3-WROOM-1-N16R8 (which has 16 MB (Quad SPI) 8 MB (Octal SPI))
Right now ESP32-C6-WROOM-1-N4  (4 MB (Quad SPI)) available on mouser, digikey, aliexpress :) 
I don´t know if the pins are compatible ...

skot, 2023-11-20:
The C6 only has a single core and it's a slower clock speed. I think we're going to stick with the S3 for now. I'll keep an eye out for a suitable RISC-V replacement in the future.  ✊

closed 2023-11-20 by skot

Lysnyt, 2023-11-20:
Hi Skot, is really necessary so much power for a serving module ? I mean, the calculation are done with The BM chip, not by ESP, am i wrong ?

ESP Announced :
ESP32-C5

    Single-core 32-bit [RISC-V](https://en.wikipedia.org/wiki/RISC-V) CPU, up to 240 MHz[[14]](https://en.wikipedia.org/wiki/ESP32#cite_note-14)
    400 KiB SRAM and 384 KiB ROM
    [IEEE 802.11ax](https://en.wikipedia.org/wiki/IEEE_802.11ax) (Wi-Fi 6) on 2.4 and 5 GHz, supporting 20 MHz bandwidth in 11ax mode, 20 or 40 MHz bandwidth in 11b/g/n mode
    Bluetooth 5 (LE)
    > 20 programmable GPIOs

Lysnyt, 2023-11-20:
Oh, i see. Thanks for explanation :)
Even without that your skilled programmers can use the S3 ESP-WIFI-MESH networking feature (or even Bluetooth mesh) to make one Bitaxe as master and other near as slave. 
-Maybe- it could harness more computing potential such inter-connected Bitaxes for splitting jobs (MESH, Webpage, network throughput, Mining admin)

eft37, 2023-11-20:
Esp32 P4 is an upcoming faster risc-v dual core board announced 9 months ago. Maybe that one?

benjamin-wilson, 2023-11-20:
At first glance looks like it could be a substitute

benjamin-wilson, 2023-11-20:
Nevermind - no wifi

benjamin-wilson, 2023-11-20:
Although it looks like they intended for you to add a  ESP32-C2 for radio hmmmm

skot, 2023-11-20:
Ooo the P4 could be good for SeedSigner

Lysnyt, 2023-11-20:
https://en.wikipedia.org/wiki/ESP32

eft37, 2023-11-21:
> Ooo the P4 could be good for SeedSigner

I learned about Seedsigner.com via this post, you think it could do that vs a raspberry pi zero?