skot/bitaxe #56: Add USB connector

SatoshyBoy, 2023-08-06:
ESP32-S3 has a USB peripheral that is capable of performing USB-JTAG. It enables anyone to program and debug by only connecting it to a computer through USB.

Lysnyt, 2023-08-08:
I was thinking the same thing, since a direct connection to a PC can be also used without ESP miner. Just like Gekkoscience USB miner.
Some time ago I suggested using ESP32-C6, which does not have USB / OTG (which is disadvantage). But its WIFI6 could be used as mesh with other Bitaxes, so only one will connect to the router with collected data from other Bitaxes.
I think the Bitaxe should be minimalistic (far it will have one BM chip). So adding a ESP32 S3 is unnecessarily or add any FTDI communication USB/TTL / 3th party CP210x or CH34x, to ESP32-C6 is complicating things. 
For first use would be good to upload firmware via programator hw and after i suggest via OTA...

SatoshyBoy, 2023-08-13:
The ESP32-C6 also comes with USB-to-JTAG functionality. Once you've experienced hardware debugging with JTAG, you'll never want to go back to printf again!

Even if you don't have plant to use the USB-to-JTAG, let these pins available. The current PCB has a bunch of free pins just hanging around.

And look, some else already had that idea.

https://twitter.com/m_mikpoly/status/1689664915034779653

Lysnyt, 2023-08-13:
Sounds good :) but over my knowledge .. Skoooott ?

skot, 2023-10-11:
USB port has been added to all boards from ultra going forward.

closed 2023-10-11 by skot