skot/bitaxe #59: ASIC chip temp not working

skot, 2023-08-21:
The pins I thought were the on-die chip temp sensor don't seem to be working. the EMC2101 will read 127ºC, the max temp all the time.

It's probably best to DNP resistors R8 and R9 to disconnect the EMC2101 from the BM1366 for now. (it will read 0ºC after that)

[image]

SatoshyBoy, 2023-09-01:
Hi Skot.

My sensor is reading the temperature. I added a pretty good thermal pad on the chip and finally got a temperature below the maximum. It is reading temperature around 63C.

Another detail that I noticed. After adding the thermal pad, the current decreases by 1A for the same hash/rate. I didn't expect that.

skot, 2023-09-01:
> My EMC2101 is reading the temperature. 

Is this on the bitaxeUltra with the BM1366?

SatoshyBoy, 2023-09-02:
Sorry, It is the BM1397. I didn't notice that this topic is for the ULTRA. But BM1397 also has an issue with the temperature. I was reading 127 every time.  

But there is something weird with BM1397, too. I can change the measured temperature by only changing the Vin. Usually, the bitaxe uses 5V, but if you try other voltage inputs, it has an effect on the temperature. It doesn't make sense because the VDD is always the same 1.4v, and power consumption (P=Vin*I) is constant.
 
Moreover, the output has a square form, which also doesn't make sense. (Yellow: temp_P. Purple: temp_N. White: temp_P - temp_N)
[image]

skot, 2023-09-02:
When you get that square wave on the ASIC die temp pins, is it also connected to the EMC2101?

Those die temp pins are connected to a diode (or transistor) on the die. To measure the temperature you put two different currents through it and the difference in forward voltage is proportional to the temperature. That would be one explanation for the square wave.

It also seems like you have a soldering problem and/or the ASIC is damaged. I have been able to read the chip temp just fine on several BM1397.

SatoshyBoy, 2023-09-03:
The EMC2101 is connected. I didn't know it has an active connection for parasitic resistance in series.

skot, 2023-11-20:
Confirmed the BM1366 does not have internal temp diode pins :(

closed 2023-11-20 by skot