skot/bitaxe #63: Replace 1.8V LDO U5 with 1.2V MCP1824T-1202E

benjamin-wilson, 2023-09-01:
voltages XP hashboard are 0.8V and 1.2V, not 0.8V and 1.8V like on the ultra

bart, 2023-09-02:
S19a Pro for example also is 1.8V

benjamin-wilson, 2023-09-02:
```
developeralgo222 — Today at 10:26 AM
Tested today. if you are using the original ultra dev board without any changes we have now tested  that if  U5 @ 1.8V LDO  <===> J-Connector (1V8)  <==>  UART  Serial ( Input 1.8V or 1.2V  both work and able to communicate to the BM1366 AG/AL chip )
But if you have it same way as in S19 XP Hashboards where its supposed to be 1.2V it does NOT work in current Ultra Dev board i.e  if  U5 @ 1.2V LDO  <===> J-Connector (1V8)  <==>  UART  Serial ( Input  1.2V  Does NOT work and will not communicate to the BM1366 AG/AL chip )
For those who planned to switch U5 to 1.2V please hold off until we can figure out why it does not work with 1.2V LDO @ U5
```
Hold off for now

skot, 2023-11-20:
the minimum supply voltage for the crystal oscillator we are using is 1.8V. if we switch to 1.2V we need to sort this out.

AxisRay, 2024-01-27:
1.2v LDO:  TLV70212QBVRQ1
1.2v crystal oscillator: SX3M25.000E20F30THN 
Test OK

DirceuPY5DK, 2024-02-18:
and if at the input of pin 14 you place a voltage divider with two resistors in series, 1k and 2k to ground, the result will be 1.2v at the input of the BM1366... the other line would have a voltage of 1.8v and at pin 14 would have 1.2v. It's just an idea.

benjamin-wilson, 2024-04-19:
Tested with a 1.2V LDO and 1.2v oscillator, it did not work. 
https://www.digikey.com/en/products/detail/onsemi/NCP114ASN120T1G/6560601
https://www.lcsc.com/product-detail/Oscillators_Shenzhen-SCTF-Elec-SX3M25-000E20F30THN_C5137262.html

AxisRay, 2024-04-21:
> Tested with a 1.2V LDO and 1.2v oscillator, it did not work. https://www.digikey.com/en/products/detail/onsemi/NCP114ASN120T1G/6560601 https://www.lcsc.com/product-detail/Oscillators_Shenzhen-SCTF-Elec-SX3M25-000E20F30THN_C5137262.html

I have built several bitaxes with 1.2V LDO and oscillator, and all of them works well.
Is it the oscillator or the ASIC that is not working correctly?
[image]

skot, 2024-04-25:
I'll try this out with some LCSC drop-in 1.2V oscillators. if it works I'd like to include in 402

skot, 2024-04-26:
Switched my v400 to the `MCP1824T-1202E` 1.2V LDO and `C5137262` 1.2V 25MHz Oscillator. seems to be working -- I'll keep an eye on it.

skot, 2024-06-07:
officially changed in 402

closed 2024-06-07 by skot