skot/bitaxe #75: Fan and heatsink specification

smartm0use, 2023-09-14:
Can you add these info to README file?

skot, 2023-09-14:
it's already there: https://github.com/skot/bitaxe#hardware

smartm0use, 2023-09-15:
There I can read that I can buy a "[40x40mm heatsink and 5V fan](https://www.aliexpress.com/item/2251832861666365.html)":

[image]

Then, I read "Noctua NF-A4x10 5V 4-Pin fan".

So, I deduce that I can buy a 4-Pin fan... but there is not 4-Pin fan in the Aliexpress link.

What's wrong? The link or the 4-Pin information?

smartm0use, 2023-09-17:
In addition, I see that it is possible to get a fan from a random seller on Aliexpress, otherwise get a Noctua fan. I saw that Aliexpress fans has more or less the same characteristics, especially in terms of rpm and db, but costs 1/3 or even 1/4.
I was wondering if I opt for a cheap fan and then maybe it spins more slowly, stops or has problems cooling, in short, if it doesn't manage to cool properly, what happens?
Does the ASIC chip burn out or does some sensor come into play so bitaxe is saved?

skot, 2023-10-11:
we are working on some firmware to shut down the ASIC and/or power in an over-temp situation. Currently there is no protection. 

This is going to be a firmware fix, so I'll close this issue.

closed 2023-10-11 by skot

smartm0use, 2023-10-11:
Ok, I see. Thank you for the information.
What about the other questions?
Is that Aliexpress link misleading?

skot, 2023-10-11:
the bitaxe is compatible with 5V fans, either 2-pin, 3-pin or 4-pin.