skot/bitaxe #88: Fix U9 footprint

benjamin-wilson, 2023-10-11:
I just got reminded that U9's footprint may be wrong. Needs investigation against the data sheet.

skot, 2023-10-11:
the little tabs coming off the GND pad were spaced too far apart. fixed in f07d92f

shufps, 2023-10-22:
You designed the footprint by yourself?

What is wrong using `Package_SON:VSON-10-1EP_3x3mm_P0.5mm_EP1.65x2.4mm_ThermalVias` ?

shufps, 2023-10-23:
Kicad has this footprint in the library:

`Package_SON:VSON-10-1EP_3x3mm_P0.5mm_EP1.65x2.4mm_ThermalVias`

skot, 2023-10-23:
wow, I didn't think to look in the KiCad library for that goofy footprint. It's pretty close, but not exactly what the TI datasheet calls out..

[image]

shufps, 2023-10-24:
Hmm idk if it matters much. Datasheet says `VSON (10) 3x3mm` and normally I just rely on this and use what Kicad gives me :man_shrugging: :grimacing:

skot, 2023-10-24:
lemme know how it goes! the TPS40305 footprint has been a major source of soldering troubles for bitaxe assemblers.