skot/bitaxe #89: Mode Pins on BM1366

shufps, 2023-10-21:
Hi,

I'm a bit confused about how the caps C4 and C16 are connected to the BM1366.

[image]

What purpose do the `mode_0`, `mode_1´ and `modeo` pins have?

And what I also wondered - where did you find this awesome information?

I searched a lot and didn't find any schematics to the BM1366 - not even in the repair guides of Bitmain.

Thx,
Thomas

skot, 2023-10-21:
The mode pins change the direction of the data pins. Ie mode1 is inputs on the left and outputs on the right. Mode0 is the opposite.

Those mode names are arbitrary of course because we don't have any schematics or pinouts.

shufps, 2023-10-21:
Thank you, this is almost what I thought for what they are :sweat_smile: 

What I still don't understand is that `mode_1` and `modeo` have caps to `0.8V`.

So I suspected there must be more going on - like extra caps for a charge-pump that uses the pins that are labeld with `modeo` and `mode_1` or so idk.

If it's not a charge pump then I absolutely have no idea what purpose the caps have :sweat_smile:

shufps, 2023-10-21:
perhaps you already saw - I did a fork for a Raspberry Pi HAT and I removed everything I thought I wouldn't need :see_no_evil: 

[image]

I also removed the two caps because I couldn't explain for what they are but at the same time I'm suspicious that I made a mistake by removing them :see_no_evil:

shufps, 2023-10-22:
I revised the schematic on my fork and added back the 1µ caps. Probably it was stupid to remove them in the first place anyways :grimacing: You seem to have real hardware for reverse engineering - I ofc don't and it was stupid to assume to know better :see_no_evil: 

Ordered my HAT and parts ... let's see how it's going :crossed_fingers:

closed 2023-10-23 by shufps

skot, 2023-10-23:
Now you have me wondering about this too. What we have is definitely working on the single-ASIC bitaxe. I'm going to look a little deeper into this for the multi-ASIC bitaxeHex.

I don't think it's a charge pump. From what I can tell it's just trying to passthrough the mode from one chip to the next in the chain.