skot/bitaxe #95: TPS40305 PGOOD signal is incorrectly pulled up to BP

skot, 2023-11-10:
[image]

according to the [TPS40305 datasheet](https://www.ti.com/lit/ds/symlink/tps40305.pdf?ts=1699556087481) BP is a 6.5V Output bypass for the internal regulator. We don't want to pullup the PGOOD open-drain signal to that as it connects to the ESP32.

the easy fix here is to just remove R11

benjamin-wilson, 2023-11-10:
Were there any consequences to having R11 populated in the past? R11 should be removed from all versions?

skot, 2023-11-10:
it doesn't _seem_ like it's been causing any problems, but pulling up to 6.5V is going to be out of spec for the ESP32. we should test this, and if it's good, then yes remove it from all versions.

skot, 2023-11-10:
I pulled off R11 on a Ultra 202 and it seems to be working fine.

skot, 2023-11-20:
fixed in 22f5b3f

closed 2023-11-20 by skot