skot/bitaxe #104: Bitaxe endless loop

RuneStone0, 2023-12-17:
I'm not sure if this is the right place to ask. I just got my bitaxe ultra, and it was working great. But when I configured it with the following:
* stratum+tcp://sha256asicboost.auto.nicehash.com:9200
* NHbaq6tWzM181eHpJeCSfPfCtnS561RSX3Ay.bitaxe

When using the username `NHbaq6tWzM181eHpJeCSfPfCtnS561RSX3Ay.bitaxe` it causes bitaxe to enter into an endless loop. It connects to wifi and immediately reboot. The web ui is only accessible for a second, and reboots before I'm able to changes to the config.

How do I flash the bitaxe ultra so I can get back to a normal state? Is there another way to fix this without flashing?

benjamin-wilson, 2023-12-17:
The URL field should only include the URL. (sha256asicboost.auto.nicehash.com) easiest way to fix it is to turn off your Wi-Fi or move out of range. It will then make the access point to connect to directly again.

tonyfung820, 2023-12-17:
Hi, if my wallet address is 1234567890. Do I have to add .bitaxe at the end of user field? Thanks!

benjamin-wilson, 2023-12-17:
It's optional. Also this issue isn't the place to ask, please use the OSMU discord.

closed 2023-12-17 by benjamin-wilson