skot/bitaxe #118: Bitaxe 1366 Severely Underhashing

555-FILK, 2024-01-19:
Hello,

I like playing around with desktop miners so I got my hands on a Bitaxe 1366 after hearing about it on various podcasts and reading about it online. I thought it would be a nice complement to my NerdMiner.

I received the 1366 and have been running it for about a week now and it has been underhashing since day one. I believe it should hash up to 550 GH/s but I am consistently getting only about 250 H/s, yes hash per second, not even KH/s.

Any tips for improving performance? Not that I expect to find a block, but if I am running it I might as well get it to high performance.

Thank you,
[image]
[image]

benjamin-wilson, 2024-01-19:
Hi, where did you get your bitaxe from? It looks like it went into overheat protection mode at some point. If you go into the settings page simply reset the frequency and voltage to the defaults.

555-FILK, 2024-01-19:
Thank you for the quick reply! I got it from D-Central. I went into settings and the Frequency was blank and the Core Voltage was blank, so I set them to their defaults, 485 and 1,200, respectively. Unit is now restarting. I will update here if it worked. Thanks again.

555-FILK, 2024-01-19:
10 minutes in and there is an obvious difference, that seems to have been the issue. However, now there is a warning about low voltage, see attached. Do I need to adjust anything in settings? Should I care? Thanks again.
[image]
[image]

benjamin-wilson, 2024-01-19:
Your chip temperature is very hot, is the fan going? What fan? Is there thermal paste?

555-FILK, 2024-01-19:
Yes the fan is going. It is set to "Invert Fan Polarity", "Automatic Fan Control" is off, and Fan Speed is 100% "S19 Simulator." The fan is the stock fan installed by D-Central, and I cannot confirm whether D-Central installed thermal paste. Attached are some photographs.
[image]
[image]
[image]

benjamin-wilson, 2024-01-19:
Can you take a picture of the back of the PCB?

555-FILK, 2024-01-19:
Here is the front and back.
[image]
[image]

benjamin-wilson, 2024-01-19:
Is it just the picture or is there some bridging on U8 and U10 between the pins?

555-FILK, 2024-01-19:
Here is a larger picture of U8 and U10. I don't see any explicit bridging but I did not look through any loops. I think it may just be the picture. I can look at it through loops if you'd like.
[image]

benjamin-wilson, 2024-01-19:
Yes, please ensure there isn't any bridging.

555-FILK, 2024-01-20:
I think you may have figured it out. After changing the Frequency and Core Voltage to defaults, it ran fine for about 2 hours, but as you noticed, it was getting very hot. After about 2 hours, it went back into overheat protection mode, the Frequency and Core Voltage values are now blank and it's back to hashing at about 200 H/s. So maybe there is bridging. I will reach out to D-Central, maybe I got a lemon. Thank you.
[image]

kakawlala, 2024-01-20:
For the Bitaxe Ultra setting, I recommend using a 40x40x20mm fan, which has greater air pressure to reduce heat.  I purchased my device from Bitcoin Merch and the fans they shipped were 20mm thick and I had to reapply thermal paste.  And updated to version 2.0.6 by itself. This version corrects the low voltage warning.  Under the preset frequency and voltage, the temperature is below 60 degrees Celsius.  I increased the frequency to a maximum of 575Hz and a maximum voltage of 1300mV.  The highest temperature will be below 65 degrees Celsius.  (If the temperature cannot be suppressed below 68 degrees Celsius, the device will easily trigger the low H/S protection) The fan will run at 4700~4900RPM during overclocking, so I replaced it with an NF-A4x20 5V PWM fan to reduce noise.  I have tried the 10mm thick fan, 10mm which should be more suitable for working on the early Bitaxe 1397.

[image]

[image]

[image]

[image]

[image]

monster4866, 2024-01-22:
i have the solution: The power supply is crap, buy a better one, you need at least 5000mv input voltage, then the fan is quiet, but take the one from D-Central, it has a higher speed and cools better, I also have a separate power supply bought

https://www.amazon.de/dp/B01HRR9GY4?psc=1&ref=ppx_yo2ov_dt_b_product_details

5v and min 4A, The power supply from them is cheap Chinese junk

LG

monster4866, 2024-01-22:
[image]

555-FILK, 2024-01-23:
Update: I put a regular house fan in front of the unit, set the Frequency and Core Voltage to defaults, and it has been running for 5+ hours and it has been running at 350 GH/s to 550 GH/s.
[image]
[image]
[image]
[image]
This is the longest I've gotten it to run at that GH/s rate. The moment I turn off the house fan off, the temperature begins rising. I'm not going to keep this setup long term, it's just a test, which confirms in my mind the stock fan is insufficient, right? [kakawlala](https://github.com/kakawlala) recommended I upgrade to a 20mm fan, would that resolve this issue?

As for the low voltage, I am still getting that issue. [kakawlala](https://github.com/kakawlala) mentioned updating the firmware. But, [monster4866](https://github.com/monster4866) recommended replacing the power supply. Any comments or recommendations on this issue?

benjamin-wilson, 2024-01-23:
I recommend you return and replace your unit, it's not operating how it should.

kakawlala, 2024-01-23:
Here are the reasons why I recommend using 20mm thick fans on BITAXE ULTRA.  I purchased mine from Bitcoin Merch and they shipped it to me with a 20mm thick fan.  

But the photos on their website are for 10mm thickness.  After receiving the item, I didn't notice it at first.  

[image]

I only found that under over-frequency and over-voltage conditions, the fan can control the temperature and prevent protection from reducing H/S, but the noise is obvious.

So I followed the recommendation and used the noctua NF-A4x10 fan, which indeed reduced the noise, but it couldn't suppress the temperature.  Only then did I discover that the shopping website was equipped with a 20mm thick fan.  After that I changed again to noctua NF-A4x20 fan.

[image]

[image]

555-FILK, 2024-01-23:
Update: I tried another 5V/4A DC PSU and that resolved the "Danger: Low Voltage" issue. It's disappointing that D-Central charges $15 for a PSU that doesn't work.

However, the alternate PSU did not resolve the overheating issues. So far, I updated the firmware and the PSU, but the heating is still an issue. I guess the next step is to upgrade the 10mm fan to a 20mm fan. Upgrading to the 40x40x20 Noctua 4pin fan next.

I want to thank everyone for their comments, suggestions, etc. What a great community.

555-FILK, 2024-01-25:
Final (?) Update: Bitaxe has been running for 12+ hours with a hash rate of over 400 GH/s. What I did to get here, with all of your help:

1. Update firmware;
2. Replace the PSU ($10.00); and
3. Replace the stock 40x40x10 fan with a 40x40x20 fan ($15.00).

I reached out to D-Central when I first posted here, but they still have not responded. It's disappointing that they charged $15 for the PSU that was junk and I paid only $10 for a replacement PSU that works. That's too bad because they had quick shipping and handling, and they included some cool stickers and 3D printing trinkets in the box it came it, which was a nice touch.

Anyways, thanks everyone for your help. Happy hashing, tick tock next block.
[image]
[image]
[image]

kakawlala, 2024-01-25:
You can cross test.  For example, use a new fan paired with the power supply the original store gave you.  The original power supply can't reach 5V, it will be around 4.85~5V, but this should make the temperature lower.  Although there was a low voltage prompt before, the monitoring value has been revised and relaxed since V2.0.5.  The main reason for low H/S protection is that the temperature detection is too high.

Next, you can set the maximum operating frequency and voltage, which will result in higher H/S.  The temperature should be below 66 degrees Celsius.  If high temperature protection occurs and H/S is reduced, thermal paste may need to be reapplied.

In the past, my experience was that when I turned the screws to replace the fan, the slight displacement caused damage to the contact surface of the thermal paste between the heat sink and the chip, and it needed to be reapplied.  After that, I used to fix the fan on the radiator first, apply thermal paste, and finally install the entire fan and radiator on the board.

[image]

CaptainCodeman, 2024-02-05:
I have 2x BitAxe's, from different vendors. One runs much cooler than the other - usually around 38 C vs 60 C for the other (which was hovering around 68 C before I took the heatsink off and re-applied the thermal paste).

One had the "Danger: low voltage" warning and seems like it consumes more power (but is the cooler one) so I replaced the PSU that came with it and it now seems to be fine. This is what I used in case it helps anyone: https://www.amazon.ca/dp/B087LY41PV?psc=1&ref=ppx_yo2ov_dt_b_product_details

Power Consumption: | 13.84 W
-- | --
Input Voltage: | 4,996 mV
Input Current: | 2,736 mA
Frequency: | 575 Mhz
Core Voltage: | 1200 mV
Measured Core Voltage: | 1194 mV
Fan Speed: | 5654 RPM
Chip Temperature: | 60 C

Power Consumption: | 15.77 W
-- | --
Input Voltage: | 4,979 mV
Input Current: | 3,168 mA
Frequency: | 575 Mhz
Core Voltage: | 1200 mV
Measured Core Voltage: | 1216 mV
Fan Speed: | 6931 RPM
Chip Temperature: | 38 C

The one that runs cooler has a 10mm height fan on it, the hotter one a 20mm fan, go figure! Both just what came with them, no fancy Noctuas (yet)

skot, 2024-02-06:
> I have 2x BitAxe's, from different vendors. One runs much cooler than the other - usually around 38 C vs 60 C for the other (which was hovering around 68 C before I took the heatsink off and re-applied the thermal paste).

What HW version are each? I'm guessing the 15W / 38C unit is actually running much hotter.

CaptainCodeman, 2024-02-06:
The one reporting the hotter temp says 201, I can't see a version on the other one though.

[image]

I only have a cursory knowledge of electronics, but thought it was odd that the one using more power would really be cooler, esp. by that much, and it did trip into the overheat protection mode a couple of times (where the frequency and power settings blank, and the fan control is cleared)

They have both been running fine since the thermal paste & new PSU though (a couple of days).

skot, 2024-02-06:
Yeah, the one that is reporting a cooler temperature is an older revision that doesn't have the sensor as close to the ASIC. They are probably both actually running at the same temp. You can put your fingers on the side of the heatsink to compare.

skot, 2024-02-06:
That could be due to the faster fan, and/or just subtle differences in the chip bin. Glad we got this sorted out!

closed 2024-02-06 by skot