skot/bitaxe #120: Modify display header pads

benjamin-wilson, 2024-01-25:
1. The display header pads are relatively small and may rip out when the display is removed. 

2. The exposed bit of the pads are too large and need to be covered with more mask. The large pads cause the header to tilt during reflow, making the screen not parallel/perpendicular. Making them the same height as the pins should make it self-align as well.

AxisRay, 2024-01-26:
I think using an FPC socket would be better? This way, we can directly attach the screen to the PCB.

skot, 2024-01-26:
Should we fix this for 205?

I'm not really a big fan of that display connector. It seems too tall and too lose, making the display look wonky often.

I'll take a look at some options.

benjamin-wilson, 2024-01-26:
The SMD version is significantly snugger in fit - this is why i'm having issues with pads ripping

benjamin-wilson, 2024-01-26:
[image]
The height seems appropriate as well.
One option to completely eliminate it would be to have staggered holes and insert it directly into the PCB
[image]

AxisRay, 2024-01-28:
my solution
I printed a 10mm*10mm*10mm cube and stick it to the pcb.

[image]

benjamin-wilson, 2024-01-30:
The more I think about it - the best part is no part. The display fits really nice and secure into those staggered holes while being easy to insert/remove. The tube the header comes in is a big pain as well.

closed 2024-01-30 by benjamin-wilson