skot/bitaxe #125: Ask for help

kakawlala, 2024-02-22:
I made bitaxeUltra 204 on PCWAY. The chip used is BM1366AL. I used esptool to flash esp-miner-factory-204-v2.0.7.bin, but it didn't work and it showed that it could only accept new jobs.  How should I measure whether there is a component failure or misalignment?  Or maybe my programming method is wrong.
The heat dissipation is also normal and there is no high temperature protection.

![P_20240222_160505](https://github.com/skot/bitaxe/assets/89348834/f4188373-73af-423b-8a28
[image]
-e88a75cf0ebf)
[image]
[image]
[image]
[image]

skot, 2024-02-22:
Everything looks good but it's not mining sounds like an ASIC issue to me. Likely the ASIC isn't soldered properly, or the chip is bad.

skot, 2024-02-22:
I'm going to close this as it isnt a bitaxe issue. Can you bring this up on the OSMU Discord? There is an invite link at bitaxe.org

Some other people have attempted PCBWay assembly with similar results.

closed 2024-02-22 by skot

kakawlala, 2024-02-22:
> 一切看起來都不錯，但對我來說，這不是挖礦，聽起來像是 ASIC 問題。可能是 ASIC 焊接不正確，或是晶片損壞。

Thanks for your reply, you are right.  After reinstalling BM1366 on the 3 previously faulty devices, there is still no hash value.  Later, after I swapped the chips of a normal and abnormal device, the results were also swapped.  Sure enough, it was a fault with BM1366.

[image]