skot/bitaxe #132: Panne Bitaxe BM1397 V3.8

Peichan83, 2024-03-06:
Bonjour,

Suite à une panne de wifi de mon opérateur et pendant une mise à jour du esp-miner.bin, mon bitaxe ne s'allume plus.

Il est détecté par mon réseau wifi, mais dès que je le sélectionne, cela me mets en échec de connexion.

Le ventilateur tourne mais l'écran lcd ne s'allume plus.

Après un passage à la loupe, aucun composant ne semble cramoisi.

Comment puis-je faire un hard reset ou comment puis-je le reconfigurer svp.

Merci d'avance.

bamatila, 2024-03-06:
Salut essaie de le reflasher via un pc, prends le bitaxe max puis après tu peux le reconfigurer en te connectant au wifi du bitaxe 

https://www.youtube.com/watch?v=e37tbrXdAtY : tuto 

https://wantclue.github.io/bitaxe-web-flasher/ web flasher

Peichan83, 2024-03-06:
Re- je veux bien mais j'ai un modèle BM1397 V3.8 sans port usb.
[image]

Avez-vous une solution de connexion?

bamatila, 2024-03-07:
AH je ne sais pas bon courage

19201003080114, 2024-03-07:
This is not a "legit" Bitaxe so ESP-Miner may not work properly with this hardware.
Since there is no USB, you need to reflash it using the connector next to the display (upper right in your picture) and use this programmer
[image]

Peichan83, 2024-03-14:
Ok J'ai commandé du coup le : 
Esp-Prog Jtag Debug & Programme téléchargement pour carte de développement Esp-Prog Jtag Debug Programme téléchargement compatible avec Esp32.

J'espère pouvoir le re flasher.

closed 2024-03-16 by benjamin-wilson

Peichan83, 2024-05-01:
Bonsoir, après avoir reçus, enfin après 1 bon mois d'attente mon esp jtag.

J'ai réussi grâce à vous à flasher mon bitax, seul problème, je n'arrive plus à le paramétrer correctement.

Pouvez-vous m'aider à nouveau?

Cordialement