skot/bitaxe #136: No Hash

BKKullerkeks, 2024-03-14:
Hi guys, I got my Bitaxe Ultra today and have everything set up so far but unfortunately I have a hash of 0. I don't know what to do, do you have any ideas? I have already reflashed the Bitaxe. 

Greetings Kullerkeks

19201003080114, 2024-03-14:
Hard to tell what could be wrong without screenshots

BKKullerkeks, 2024-03-14:
[image]

BKKullerkeks, 2024-03-14:
[image]

19201003080114, 2024-03-14:
Wrong pool address and port

https://solo.ckpool.org/

BKKullerkeks, 2024-03-14:
still the same 0 Hash

19201003080114, 2024-03-14:
Address should be "solo.ckpool.org", NOT "de.ckpool.org"
Port should be "3333" unless you want high diff then use "4334"

BKKullerkeks, 2024-03-14:
its the same 0 hash

19201003080114, 2024-03-14:
Did you restart it after saving the changes?

BKKullerkeks, 2024-03-14:
yes, i think it is maybe a firewall problem

BKKullerkeks, 2024-03-14:
[image]

he4dhuNTeR, 2024-03-14:
Wrote your wallet address correctly? Ck checks this at connecting. Also already tried another pool like VKpool or publicpool?

BKKullerkeks, 2024-03-14:
yes i tried all

he4dhuNTeR, 2024-03-14:
Just as an offer if you have Telegram, you can search for "BITAXE GERMAN" group, we have some real cracks there for hardware and software, maybe they can help you immediately. And communication is easier on TG.

BKKullerkeks, 2024-03-14:
thx thats a god idea

closed 2024-03-16 by benjamin-wilson

BBRENESS, 2024-08-25:
did you ever find the issue?

19201003080114, 2024-08-26:
If you use high diff port, it is normal that it is not working.