skot/bitaxe #141: V205 type-c cannot connect

kakawlala, 2024-03-29:
I created version 205. The obvious differences between this version and version 204 are: D1 is removed, U10 is added, and type-c is changed to the right.  I pressed the boot button and then connected type-c. The computer did not detect the device. Which general direction should I check for the problem?

I have versions 201, 204, 400.  They have the same type-c port on the left side and can detect the device and flash normally.

[image]

skot, 2024-03-29:
v205 has been tested and is working well. (And those changes have been pulled into v401, fwiw)

To place the ESP32 into bootloader "DFU" mode, hold the boot button while you press and release the RESET button. It's probably not enough to just press BOOT

I have noticed some flakeiness with this. If it doesn't work the first time try it again.

benjamin-wilson, 2024-03-29:
Power from the USB has also been removed, please power the device with a barrel plug when trying to connect via USB. Closing as non-issue. If you need more help, please is the discord.