skot/bitaxe #178: R2,R3 and R4

Alexgtc, 2024-07-28:
In the gabers come the resistors r2,r3 and r4 but the list of materials do not come. Do you know if they are necessary and if so, what value do they have to have

19201003080114, 2024-07-28:
If not in BOM, assume DNP

Alexgtc, 2024-07-28:
What is DNP? Do you mean that if it's not on the list, don't put them? Would it work well then without putting those resistances?

19201003080114, 2024-07-28:
DNP is Do No Populate. You leave them empty. 
If you want to be sure, check the schematics.

Alexgtc, 2024-07-28:
Where can you find the diagrams of the supra 402 bitaxe plate?

19201003080114, 2024-07-28:
https://github.com/skot/bitaxe/tree/supra-402

402 is not ready. Yes 401

Alexgtc, 2024-07-28:
What is missing from the plate to be finished? With the current bom list and the current gabers it doesn't work?

skot, 2024-07-28:
Please only use issues for bugs. For assembly help please ask on the OSMU Discord

closed 2024-07-28 by skot