skot/bitaxe #181: "DEVICE OVERHEATED" message retained on display after updating settings, cooling, and restarting

satscoffee, 2024-08-21:
My device overheated...unsure why but it did but seemed to shut down before causing damage.

 I was using the default settings but was moving the device around to stop the fan from vibrating. Regardless, it overheated and stopped hashing. I went to the webUI as directed and updated the settings back to default and saved. I unplugged the device and let it cool down to room temperature. 

When plugging Bitaxe back in and starting up, it was hashing correctly but the display still toggles back and forth of showing the hash rate, but also saying "DEVICE OVERHEATED" for a bit before toggling back to the hash. Below that, the message of "Please check webUI for more information" remains on the display continuously.

Current Setup:
Device: Bitaxe 401 Supra
Latest Release: v2.1.10
Frequency: 490
Voltage: 1166
Current Hash: 667.61 Gh/s
Power: 13.1W
Voltage: 5V
ASIC Voltage Requested: 1.17V
ASIC Temperature: 55C
Fan: 56%
ASIC Frequency: 490Mhz
ASIC Voltage Measured: 1.19V

Still haven't mined any blocks. Should I add this as another issue haha.

19201003080114, 2024-08-26:
Watch the WantClue video on YouTube on how to reset overheat message.

OsO-1973, 2024-08-26:
Same problem here, can u put up a link as I don’t find the video
Thx in advance

benjamin-wilson, 2024-08-26:
Please reach out to the seller for support, GitHub is for issues with the project and not after sales support.

closed 2024-08-26 by benjamin-wilson

OsO-1973, 2024-08-26:
Thx Benjamin for the swift answer, it is about the project imho as there was no explanation how to reset it
Found it myself
https://youtu.be/wlHHc9InK6w?si=Jp54Zd3egflS9TRU  at exactly 07:30 min how to reset the message as there is no written documentation
Now it is completed
Cheers
OsO

satscoffee, 2024-08-26:
After watching the video, there is definitely a bug as my webUI did not/does not have the "Disable overheat mode" checkbox. 

I just forced the Bitaxe to overheat again by lowering the fan and confirmed again there is not a checkbox after reviewing the webUI.

Running Latest Release: v2.1.10 but it's possible the overheat initially happened on the prior version, then I upgraded to the new firmware before resetting the frequency and voltage.

OsO-1973, 2024-08-26:
You have to update the webgui (download www.bin) then throttle the fan , it will overheat > then it appears like in the video and then the checkmark disappears.
Shd be solved then 
[image]

19201003080114, 2024-08-26:
Do you have latest www.bin installed?

satscoffee, 2024-08-26:
Thank you! Done. It worked