skot/bitaxe #183: Supra stops hashing

M31andromeda, 2024-09-08:
So I've built 6 Supras so far. While 4 of them are working as expected, 2 of them always stop hashing shortly after power-up. WiFi connection is there. I've also replaced the ASIC on one unit but the issue persists. Not sure where else I should look. Any ideas? 

[image]

M31andromeda, 2024-09-08:
Running longer now but operation is not looking right...

[image]

Screenshot of another Supra working as it should:

[image]

benjamin-wilson, 2024-09-08:
On your second post that looks like normal operation, averaging to the expected hashrate.

closed 2024-09-08 by benjamin-wilson