skot/bitaxe #192: Bitaxe supra self test monitor fail

MabsHA, 2024-10-15:
my bitaxe wasn’t hashing anymore after i installed the 2.2.2 firmware i downgraded to the 2.1.10 still didn’t work. i reflashed with the webflasher New Bitaxe Supra: Board Version 401 (factory)V2.1.10 but now i get message (monitor fail) after self test with serial readings i get this message (self_test: INA260 test failed!) ill ad log download.

any ideas?
[esp-web-tools-logs.1.txt](https://github.com/user-attachments/files/17379525/esp-web-tools-logs.1.txt)

19201003080114, 2024-10-15:
Likely you have a hardware failure, and not software related.

MabsHA, 2024-10-15:
> Likely you have a hardware failure, and not software related. 

Can it be fixed?

MabsHA, 2024-10-15:
> > Can it be fixed?
> 
> 
> 
> Replace U8, INA260

That's only €2,50 I'll try

skot, 2024-10-15:
Github issues are not for build issues. please use OSMU.

closed 2024-10-15 by skot

MabsHA, 2024-10-15:
Where can I find that?

skot, 2024-10-15:
https://discord.gg/osmu

MabsHA, 2024-10-15:
And I didn't know if it was firmware or hardware sorry