skot/bitaxe #194: My new Bitaxe Gamma 0.00 H/s

JCF711, 2024-10-22:
Try with diferent internet connections, and diferents Pools.

₿ (1137946) stratum_task: Clean Jobs: clearing queue
₿ (1138956) stratum_task: Connecting to: stratum+tcp://public-pool.io:21496 (172.234.17.37)
₿ (1138956) stratum_task: Socket created, connecting to 172.234.17.37:21496
₿ (1139076) stratum_api: Resetting stratum uid
₿ (1139076) stratum_task: Clean Jobs: clearing queue
₿ (1139076) stratum_api: tx: {"id": 1, "method": "mining.subscribe", "params": ["bitaxe/BM1370/v2.3.0"]}
₿ (1139086) stratum_api: tx: {"id": 2, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}
₿ (1139106) stratum_api: tx: {"id": 3, "method": "mining.authorize", "params": ["xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx", "x"]}
₿ (1139116) stratum_api: tx: {"id": 4, "method": "mining.suggest_difficulty", "params": [1000]}
₿ ### (1139286) **stratum_task: rx: {"id":1,"error":null,"result":[[["mining.notify","57601348"]],"57601348",4]}**
₿ (1139286) stratum_api: extranonce_str: 57601348
₿ (1139296) stratum_api: extranonce_2_len: 4
₿ (1139376) **stratum_task: rx: {"id":2,"error":null,"result":{"version-rolling":true,"version-rolling.mask":"1fffe000"}}**
₿ (1139376) stratum_api: Set version mask: 1fffe000
₿ (1139376) stratum_task: Set version mask: 1fffe000
₿ (1139386) **stratum_task: rx: {"id":3,"error":null,"result":true}**
₿ (1139396) stratum_task: setup message accepted
₿ (1139396) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[1000]}
₿ (1139416) stratum_task: rx: {"id":null,"method":"mining.notify","params"

Result, 0.00 Hash Rate

skot, 2024-10-22:
Despite what it looks like, there are no errors here. Please contact the seller, or use OSMU (discord.gg/osmu) for support.

closed 2024-10-22 by skot

billytaylor, 2024-10-24:
@JCF711 I noticed that my Gamma was 0.00H/s too after I had updated the asic freq to '525 (default)' in the web UI. I set it down to 490 and it's hashing happily again now. HTH, cheers

JCF711, 2024-10-24:
In the end it was an overheating problem. When the protection was triggered the voltage was set to 1000, and I set the frequency to 525Mhz (by default), and so it couldn't mine. The thing is that I can only use it at 400Mhz (voltage >1050) so it doesn't overheat (and even then it stops every 3-4 hours). I'm going to change the thermal paste for a specific one for 'ASIC' and I'll see if I can solve the problem.