skot/bitaxe #196: GTIN-13 Global Trade Item Number BitAxe Gamma

BdGBertdeGroot, 2024-11-01:
We couldn't find an officially registered GTIN-13 number which is necessary to be able to sell the Bitaxe Gamma

We've officially registered it with the GS1.

The code is
8720892478016

We will pay the €60,- per year to keep the code officially registered.

If there are more official products going to be mass produced that need a barcode for merchants/ global offering please let us know via Bitcoinbrabant.com

Attached the codes and documents, free for everyone to use.

[8720892478016.pdf](https://github.com/user-attachments/files/17600154/8720892478016.pdf)
[image]
[image]

skot, 2024-11-01:
Very cool! Thanks for setting this up. @WantClue should we add this to the wiki?

benjamin-wilson, 2024-11-01:
We should probably add it to the repo too

WantClue, 2024-11-01:
Yes doing it rn!

skot, 2024-11-01:
@BdGBertdeGroot do you know what parts of the world require a GTIN-13 number?

WantClue, 2024-11-01:
Done!