skot/bitaxe #197: Internal voltage domain bypass caps

kingovchouffe, 2024-11-10:
Hello,

I'm a hardware design engineer, and the part in the schematic of the bypass caps for the internals voltages domains makes absolutely no sense electrically speaking. 

Do you have documentation about the miner chip that specified this arrangement? 

I am willing to help with the hardware design, this project looks very cool !

skot, 2024-11-11:
Hey, welcome!

There is no documentation... All we have are some cryptic "repair guides" and closely looking at existing hashboards.

kingovchouffe, 2024-11-11:
I tried to look at some documentation about those ASIC but there is not much appart those repair guides which you mentioned. My theory is that their retro engineering is wrong in the symbol pin out, but someone it works. But bypassing two DC voltages with capacitors makes nonsense to me. I'll try to find some better picture of Asics board

tkircher, 2024-11-27:
According to this, there should be no connections to VDD1, VDD2, or VDD3 in a single chip configuration:

https://www.zeusbtc.com/articles/information/4915-antminer-s19-xp-hydro-hash-board-repair-guide

skot, 2024-11-27:
Are you referring to this diagram? [image]

That appears to be a higher level diagram. If you look at the actual hashboard you'll see each internal voltage domain has bypass capacitors.

tkircher, 2024-11-27:
You're right, I didn't recognize that at first. I found this for the BM1397 that has the same basic design:
[image]
And this:
[image]
@kingovchouffe is right that this is a pretty unusual arrangement.