skot/bitaxe #198: SELF TEST FAIL, INCORRECT CORE VOLTAGE

sospcie, 2024-11-10:
Hi All,

Finally I was able to build my Bitaxe Gamma device with BM1370BC. Was not easy but was fun. :)

Flashed the firmware, display working, fan spinning, logs display the chip detected, but then I get this message on screen and in the terminal as well

self_test: SELF TEST FAIL, INCORRECT CORE VOLTAGE

Does anyone can give me a hint which of the chip responsible for the CORE VOLTAGE?

Thank you

19201003080114, 2024-11-11:
[image]

benjamin-wilson, 2024-11-11:
Github is not the right place for troubleshooting individual issues, this is for fundamental issues with the project.

Visit https://discord.gg/osmu for help.

closed 2024-11-11 by benjamin-wilson