skot/bitaxe #203: Onboarding pain: empty frequency value

alltheseas, 2024-12-12:
_What happpened_

New 601 gamma bitaxe was not hashing. After too many moments of trying to figure out what is broken, it turned out the 'Frequency' value was null. I did not set this to null. 

It could be that overheating protection triggered this null value, or value was null by default. 

_Suggestion_

In order to reduce onboarding friction, consider including a UI signifier that clearly communicates to a non-technical newbie that a Frequency value is required in order for the miner to hash.

### Mockup

[image]

mrv777, 2024-12-12:
Hmm, I'm not sure all of the cases of this or how it got to null, but we could maybe add an alert on the dashboard if the frequency is below maybe 400 (that includes null)

alltheseas, 2024-12-12:
That would have prevented me one hour of head banging, sounds great.

Out of curiosity, what unhappy paths would cause AxeOS to set a frequency value 400 or lower (non-null)?

eandersson, 2024-12-12:
It's likely not set to null, but rather 50 does not exist in the drop down so it does not know what to map it too. The same is seen when overclocking as well.

alltheseas, 2024-12-12:
> It's likely not set to null, but rather 50 does not exist in the drop down so it does not know what to map it too. The same is seen when overclocking as well.

Why is that hashrate was zero? Does anything below some value (e.g. 400) not produce hashrate?

eandersson, 2024-12-12:
> > It's likely not set to null, but rather 50 does not exist in the drop down so it does not know what to map it too. The same is seen when overclocking as well.
> 
> Why is that hashrate was zero? Does anything below some value (e.g. 400) not produce hashrate?

It should still produce some hashrate. If it wasn't something else was likely wrong as well (e.g. statup process failed), and I don't believe the code could handle a zero or null value without crashing at startup.