skot/bitaxe #207: AxeOS incorrectly states restart is needed to change hostname

alltheseas, 2024-12-13:
_what happens_
1) change hostname
2) AxeOS displays restart required for changes to take effect
3) I observe that the hostname name is updated in Swarm view without a restart

[image]

_suggestion_

Consider removing the "restart needed" in case of hostname change

skot, 2024-12-13:
I think you'll find that if you change the hostname and the WiFi details you _will_ need a restart for either of them to take effect..

alltheseas, 2024-12-13:
> I think you'll find that if you change the hostname and the WiFi details you will need a restart for either of them to take effect..

Does the following logic work:

1. if hostname only is changed, then do not prompt for a restart
2. if wifi info is changed, then prompt for a restart