skot/bitaxe #215: No Display & No WiFi on Bitaxe Supra 401 – I2C Errors

Pilotkosinus, 2025-02-01:
I'm building a Bitaxe Supra 401, and it's my first time soldering SMD components. I just finished assembling everything today. The WebFlasher worked without issues, but I’m not getting anything on the OLED display, and I don’t see a WiFi network.

I checked the serial monitor and found these errors:

```
E (1144) i2c.master: I2C transaction timeout detected
E (1234) display: No display found.
W (1234) self_test: DISPLAY not found!
E (1324) i2c_bitaxe: Device EMC2101 (0x4c)
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE)

```
Full serial log: https://pastebin.com/TQHnLJFt

What should I check to troubleshoot this issue? Thanks for your help!

Pilotkosinus, 2025-02-01:
I can also add that I have checked the OLED and the EMC2101 with the multimeter and both are at 3.3v.

skot, 2025-02-02:
please ask on OSMU (discord.gg/omsu) for build help.

closed 2025-02-02 by skot