skot/bitaxe #218: Change the algorithm ?

Uefi1, 2025-02-25:
Hi! Is it possible to change the outdated SHA-256 algorithm to at least SHA-256d in the firmware ? Most decent pools have been using SHA-256d for a long time now! It would be even better if there was an option to switch the algorithm to Scrypt or RandomX for altcoins.

closed 2025-02-25 by skot

Uefi1, 2025-02-25:
> it is _impossible_ to change the algorithm on an ASIC.

Why is it impossible on ESP32? There are plenty of libraries with different algorithms, just change the code.

MyOwn2C, 2025-02-25:
> Why is it impossible on ESP32 there are plenty of libraries with different algorithms, just change the code.

Because ESP32 is a multi purpose processor, not a task specific ASIC. 
ESP32 can do many things. 
BM ASIC can only do one thing.

Uefi1, 2025-02-25:
> > Why is it impossible on ESP32 there are plenty of libraries with different algorithms, just change the code.
> 
> Because ESP32 is a multi purpose processor, not a task specific ASIC. ESP32 can do many things. BM ASIC can only do one thing.

I have already seen that the bm1366 chip worked on altcoin algorithms and quite successfully !

MyOwn2C, 2025-02-25:
> I have already seen that the bm1366 chip worked on altcoin algorithms and quite successfully !

There are many altcpins using SHA256 so BM asics will work on them. 
If you are referring to UnMineable, they only pay you in the coins you want. You still mine the algo of your ASIC.

MyOwn2C, 2025-02-25:
> No, I don’t mean anything like Unmineable. Just pure Scrypt hashing directly from the ASIC instead of SHA-256.

There are ASICS that can do Scrypt. 
There are ASICS that can do SHA256. 
If there is one that can do both, they are general purpose ASICs but they are terribly slow. 
I think you have wrong understanding of what an ASIC is.

Uefi1, 2025-02-25:
> > No, I don’t mean anything like Unmineable. Just pure Scrypt hashing directly from the ASIC instead of SHA-256.
> 
> There are ASICS that can do Scrypt. There are ASICS that can do SHA256. If there is one that can do both, they are general purpose ASICs but they are terribly slow. I think you have wrong understanding of what an ASIC is. I think you also have wrong understanding of SHA256 and Scrypt.

Look, in BitAxe, there is a BM1366 chip and an ESP32, which contains the firmware that controls the BM1366 chip. So, in the firmware, any algorithm can be implemented.

MyOwn2C, 2025-02-25:
> Look, in BitAxe, there is a BM1366 chip and an ESP32, which contains the firmware that controls the BM1366 chip. So, in the firmware, any algorithm can be implemented.

Your understanding of the system architecture of Bitaxe is completely off. 

ESP32 is a processor. It sends work to BM1366 to process. 

BM1366 can only do SHA256. It cannot do anything else. 

This discussion is completely pointless until you understand what ESP32, BM ASICS and firmware actually can or cannot do. 

If you still think it is possible, you are welcome to take the ESP-miner source code and try it itself. 

Failure is a very good teacher.