skot/bitaxe #221: Bitaxe 204 - Input Voltage and Power too low. No longer hashing

aindriu18, 2025-03-01:
Last night (Feb 28), I noticed the fan on my Bitaxe made a strange noise and suddenly started to get quiet. On further investigation this morning, I noticed it has practically stopped. The fan is operational, but barely. On checking AxeOS, both power and input voltage are extremely low. Both frequency and core voltage are set to default and have been for some time. I've also made sure since day one to use the correct plug for the device. 

The device is over a year old. Is it a case that it has just malfunctioned? I've tried several basic steps such as resetting it, turning it off and on, changing frequency and core voltage settings, but nothing has kicked it into gear. As of this morning, I've also updated the firmware to v2.5.0. I believe it was v2.2.0 before that.

I also notice some strange UI from AxeOS which I have included here. Note how the input voltage is in the correct location. 

[image]

skot, 2025-03-01:
It looks like a few things have happened;

- your fan died.
- that probably caused the bitaxe to go into overheat mode.
- you haven't updated www.bin (AxeOS) so you're not seeing the warning.

Bitaxe issues is not the place for troubleshooting. Please reach out on the OSMU Discord (https://discord.gg/osmu) and we can help you get this fixed.

closed 2025-03-01 by skot

ST1PRO, 2025-03-02:
Hi, not sure if this happened to anyone else. But I plunged into a power strip and got a low voltage warning then later unplugged and plugged directly into the wall and the voltage warning went away.